DiedB/Homey-SolarPanels #262: Enphase no longer working after update v5.2.0

FrankSmeets01, 2023-06-10:
After today’s update to v5.2.0, my Enphase gateway is no longer working in Homey. MessagePad in device in Homey reports “Enphase Envoy can no longer be discovered om your Network”. The solar system is working fine and no issues in the Enphase app. 

Can this please be resolved?

Diagnostic report ID: d1cfa21a-d023-4c18-8e9d-345a9fdbcdc8

DiedB, 2023-06-13:
That's weird, nothing changed in the Envoy code, so the issue should not be within the update. The app probably restarted because of the update and then started searching for your Envoy again which failed. Did you change anything to your network configuration and is the Envoy on the same subnet, behind the same router as your Homey?

FrankSmeets01, 2023-06-14:
You were right. It's not due to the new version. I see that it started working again after some time when I restarted the app again. I didn't change anything to my network setup. Thx anyway.
And thx for this great app!

closed 2023-06-14 by FrankSmeets01