DiedB/Homey-SolarPanels #265: Zonnepanelen app 2023

SWAHO1, 2023-06-22:
Hoi, de Zonnepanelen app op de Homey Pro 2019 is super.. echter nu overgestapt op de Homey Pro 2023 maar de app is niet compatibel. Simpele vraag.. komt er een nieuwe app die wel werkt op de Homey Pro 2023 met Solis Ginlong ondersteuning? Alvast bedankt voor het beantwoorden 😁

SWAHO1, 2023-06-23:
Ik heb nu de Homey Pro 2019 draaien met alleen de Zonnepanelen app en die zend webhooks naar de Homey Pro 2023 zodat ik nog wel de app kan gebruiken maar het is wel een extra apparaat voor alleen een app 😁 maar werkt wel..

wieooit, 2023-06-25:
Ik ben niet van plan beide te laten draaien dus als ook deze app niet geüpdatet wordt of vervangen door een ander begin ik te twijfelen aan de nieuwe Homey.

wieooit, 2023-06-28:
Inderdaad.
De testversie werkt bij mij vooralsnog PRIMA!
Dank je wel.

SWAHO1, 2023-06-28:
> De test versie van de zonnepanelen app werkt al wel op de HP2023. zie https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/ Let wel op dat dit een test betreft er kunnen dus nog foutjes in zitten.

Helaas geen support voor Sollis ! En op de Homey 2019 werkt hij wel met Solis.

SWAHO1, 2023-06-28:
> Inderdaad. De testversie werkt bij mij vooralsnog PRIMA! Dank je wel.

Solis wordt niet ondersteund dus ik heb er niets aan.. @DiedB komt er weer ondersteuning voor Solis? Want ik stuur nu webhooks van mijn oude homey 2019 naar de 2023.. alleen voor deze app.. thx.

Jaccoes, 2023-06-29:
Nieuwe homey gebruiker hier. Pro-2023 met Solis inverter. Als die ondersteuning er zou komen zou dat heel mooi zijn!

JohanTim, 2023-07-04:
LS is deze variant van de app ook nog geschikt voor HP 2019 en andere?

DiedB, 2023-07-04:
App is backwards compatible naar Homey Pro 2019, maar verliest wel support voor een aantal inverters.

Solis wil ik overwegen te implementeren, maar nu de app 'verified' is, moet ie aan bepaalde eisen voldoen als het gaat om gebruiksgemak in de pairing flow. Een support request bij Solis indienen, en vervolgens een API key kopiëren en plakken zit wel een beetje op het randje.

Dit gaat overigens niet alleen over de regels die Athom stelt, maar ook vooral om mijn tijd. Alles wat niet heel erg gebruiksvriendelijk is en voor iedereen werkt, levert een aardige bak issues op in deze repository. Ik heb helaas geen tijd meer om daar bovenop te blijven zitten.

SWAHO1, 2023-07-04:
> App is backwards compatible naar Homey Pro 2019, maar verliest wel support voor een aantal inverters.
>
> Solis wil ik overwegen te implementeren, maar nu de app 'verified' is, moet ie aan bepaalde [...]

SWAHO1, 2023-07-04:
"Solis wil ik overwegen te implementeren" .. je zou er een hoop mensen blij mee maken. Even een nieuwe inverter kopen is niet zo duurzaam. Ik snap je tijd issue... wij hebben hoop!

DiedB, 2023-07-04:
Snap ik! Ligt echter niet alleen bij mij, maar ook vooral bij consument en fabrikant. De inverter vervangen voor cloud features hoeft gelukkig nooit. Als consument heb je m.i. op dit moment twee keuzes:
- Een goedkope Chinese inverter kopen en alle cloud functionality afkoppelen, en daarnaast een HomeWizard meter in je groepenkast hangen (laat de inverter bezig met zijn core functie en borg smart home koppeling elders)
- Een A-merk kopen (Enphase, SolarEdge, SMA, Fronius, Tigo, ...) en profiteren van cloud-functionaliteit

Deze app focust zich vooral op de laatste categorie, omdat cloud functionaliteit robuust is, generiek is voor alle modellen inverters, en gemakkelijk in gebruik. Uitzonderingen bestaan - ik heb Growatt geimplementeerd omdat ik er zelf ook eentje heb hangen en op die manier relatief makkelijk support kan leveren. Echter kan die integratie zomaar ophouden met bestaan omdat ie meelift op een niet gedocumenteerde API. Wellicht dat die uitzondering er inderdaad ook komt voor Solis, hierboven is te zien dat er in ieder geval een hoop vraag naar is.

SWAHO1, 2023-07-04:
Ik stuur nu een webhook vanaf mijn oude homey 2019 met je app naar de nieuwe homey 2023 .. want je oude app op 2019 werkt nog wel met mijn Solis inverter.. dus ik wacht geduldig.. 6 jaar geleden was ik nog niet bezig met de inverter en wat ik daar mee moest gaan doen.. als deze stuk gaat dan gaan we idd even kijken naar een ander merk. Maar om het geld er echt uit te halen.. hoop ik natuurlijk dat deze nog even mee gaat 😁

SWAHO1, 2023-07-04:
Is er geen link naar de oude apps zodat ik het weer werkend kan maken?

DiedB, 2023-07-04:
Als je niet technisch onderlegd bent is dat best lastig. Als je op korte termijn deze verbinding nodig hebt, raad ik aan om https://www.homewizard.com/nl/shop/wi-fi-kwh-meter-1-fase (of 3-phase, afhankelijk van je inverter) te laten installeren. Laat die inverter lekker hangen tot ie kapot is en doen waar ie voor gemaakt is - inverten.

De [beschrijving van de Home Assistant app](https://github.com/hultenvp/solis-sensor) biedt niet heel veel hoop - het verkrijgen van een API key is geen simpel proces, en de API is erg instabiel. Mocht dat in de toekomst veranderen hoor ik het graag, het kan altijd nog geïmplementeerd worden.

SWAHO1, 2023-07-05:
Ik ben geen programmeur maar best technisch, alleen wil ik zo weinig mogelijk tijd investeren om alles te automatiseren. Homey beloofd dat iedereen zijn huis kan automatiseren doordat het toegankelijk is. Het heeft altijd goed gewerkt op de Homey 2019 pro (tot eer-gister). Solis is sinds een paar maanden helemaal gemigreerd naar een nieuwe cloud en een nieuwe app. Ik wil je best toegang geven tot de cloud. Ik gebruikte je app voor realtime opbrengsten te koppelen aan o.a. mijn zonnescherm. Ik ondervond geen problemen tot eer-gister de app was geüpdatet. Als je mij een link kan geven met de oude app zodat ik hem weer kan installeren zou mij dat een hele hoop geld schelen en ben ik weer even geholpen. @DiedB Natuurlijk waarderen wij je werk en inzet voor alles, alleen is het zo frustrerend dat wij (niet programmeurs) zo afhankelijk zijn van jullie met Homey. 😁

RoadXY, 2023-07-05:
Ik heb naast Homey ook Home Assistant draaien en daar ben ik alle instellingen en historie al een keer kwijt geweest door een database corruptie.
Dit bleek al een tijd in mijn database te zitten waardoor ook mijn historische backups op Google Drive ook waardeloos zijn.
Ook de officiële Enphase integratie werkt niet meer waardoor je via een community store (HACS) een niet officiële versie moet installeren.
Automatisering vele malen krachtiger dan die van Homey echter ook veel, veel moeilijker om te begrijpen.
En in mijn werk als extern systeem beheerder en als hobby programmeur zie ik een hoop software dus ik vind dat ik mag zeggen dat ik er goed gevoel voor heb.
Dit wordt ook onderschreven in de diverse YouTube tutorials.

Wat ik hiermee wil zeggen is dat ook Home Assistant niet perfect is.
Allen hebben zo hun voor- en nadelen.

SWAHO1, 2023-07-05:
> Ik heb naast Homey ook Home Assistant draaien en daar ben ik alle instellingen en historie al een keer kwijt geweest door een database corruptie. Dit bleek al een tijd in mijn database te zitten [...]

Ja natuurlijk heeft alles nadelen en voordelen maar van een apparaat van ruim 400 euro (Homey) of Home Assistant dat draait op een afgedankte intel nuc ofzo.. Ik wilde voor gemak gaan.. maar daar heb ik nu spijt van. Je bent zo (we zijn heel blij met de mensen die zoveel tijd er in stoppen hoor!!! Programmeurs top!!!) Afhankelijk van de programmeurs. Want van Athom hoef je het niet te verwachten blijkbaar.

RoadXY, 2023-07-05:
Maar bij Home Assistant ben je ook afhankelijk van programeurs die het in hun vrije tijd doen...

SWAHO1, 2023-07-05:
> Maar bij Home Assistant ben je ook afhankelijk van programeurs die het in hun vrije tijd doen...
Dat is waar, maar dan betaal je geen 400 euro voor een apparaat.. ;-)

SWAHO1, 2023-07-07:
Ik ben dan niet handig met programmeren maar wel handig met hardware. Ik heb een Eastron slimme kWh geïnstalleerd. Met de Homewizard app en de power of the hour app. Dus ik hoef nu niet te wachten op de zonnepanelen app wanneer deze Solis gaat ondersteunen. 
[image]

closed 2023-07-10 by DiedB