DiedB/Homey-SolarPanels #268: Zonnepanelen app

thomasc1987, 2023-07-03:
Sinds de zonnepanelen app vandaag is geüpdatet naar versie 5.2.0 werkt de app niet meer. Hij geeft telkens de melding “apparaat onbeschikbaar, an unknown error occured while fetching inventer data”. Iemand een idee wat er aan de hand is?

FonsdeRond, 2023-07-04:
Ik heb hetzelfde probleem als thomasc1987. Apparaat verwijderd en geprobeerd opnieuw aan te maken (op een Homey). Nu wordt de API niet meer geaccepteerd. Echter de API van  Solax Cloud Systems wordt niet meer geaccepteerd, omdat het indeling niet juist zou zijn.

thomasc1987, 2023-07-04:
Ik heb een solar edge omvormer. Ik heb nog niks verwijderd o.i.d. want ik weet eerlijk gezegd niet hoe ik alles opnieuw zou moeten instellen. Ik heb het homey systeem namelijk overgenomen van de vorige bewoner. 

Als ik in de app bij instellingen kijk dan zie ik bij de API key alleen maar zwarte bolletjes staan (32 stuks), dus ik twijfel of de API key wel is ingevuld. Voorheen was de key zichtbaar, maar ik weet niet zeker of dit is veranderd met de update van de homey app van ongeveer een maand geleden.

Als ik de zonnepannelen app (via settings) opnieuw opstart, dan werkt de app ongeveer een minuut en zie ik ook actuele data van het opgewekte vermogen. Dus het lijkt er op dat er wel nog data binnenkomt en de api key dus nog geldig is. Maar om de een of andere reden stopt de app vervolgens binnen 1 minuut met werken en verschijnt er een rode gevarendriehoek met uitroepteken en kunnen geen data meer bekeken worden. Die data lijken wel nog altijd zichtbaar in homey Insight. 

Ik denk dat ik vooralsnog even ga wachten op een update die hopelijk snel volgt en dit probleem verhelpt.

DiedB, 2023-07-04:
Dat is vreemd, heb je toegang tot het SolarEdge account? Daar zou je een nieuwe API key kunnen maken en de pairing flow kunnen doorlopen.

FonsdeRond, 2023-07-04:
Sorry, ik denk dat de Zonnepanelen app een Solax (met een X) niet meer ondersteund. Ik zag dat er nu een App is voor Solax Converters. Die heb ik geinstalleerd en de API ingevuld. Dat werd geaccepteerd, maar ik krijg nog geen waarden binnen. Nog ff zoeken dus.

thomasc1987, 2023-07-04:
Fons: Maar Solax is iets anders dan Solar Edge toch?

BasPost, 2023-07-13:
Hoi, ik heb ook sinds de update zelfde problemen met de app.

hierbij diagnostisch rapport code:
d3c12d6c-548a-4856-9479-fae551b23142

thomasc1987, 2023-08-23:
Inmiddels zijn er een aantal updates geweest maar helaas werkt de app nog steeds niet. De flows die eraan gekoppeld zijn werken opmerkelijk genoeg nog wel, dus data komt blijkbaar wel nog binnen