DiedB/Homey-SolarPanels #269: People are still using Omnik inverters

remcodk, 2023-07-04:
The Omnik support is removed from the app.

But I used it every day. It worked great. I would love to see it back.

DiedB, 2023-07-04:
I am aware. Some things have changed, which were reason to the Omnik integration being deprecated in the new app version:
- I have significantly less time for supporting this app
- The app is now 'verified' and works on Homey Cloud
- The app has been rebuilt from the ground up to modernise the code base and achieve the former
- I can only update the app as a whole, not just specific drivers

As for Omnik specifically, the driver only worked with select models of inverters, but many of them didn't expose the necessary interface. This caused confusion for users with Omnik inverters for whom the app simply did not work. For users with working Omnik inverters, the pairing process was not too easy to follow, and would not meet Athom's guidelines for user friendliness. Additionally, this resulted in lots and lots of issues being made in this repository.

If Omnik provided a standardized, secure and easy to use interface to connect to smart home hubs, I would happily keep supporting it. However, this is not the case and I would recommend you measuring solar power output using something that does tick these boxes. An option would be to install a HomeWizard kWh meter in your panel.

Alternatively, you could maybe request a community developer to build the Omnik integration as its own app. You have my permission to copy and paste my Omnik implementation into a new Homey app.

closed 2023-07-10 by DiedB

lucasvdh, 2023-08-08:
@remcodk I've just added the finishing touches to my pull request https://github.com/DiedB/Homey-SolarPanels/pull/245 which should hopefully make the Omnik driver available again if it meets @DiedB's ands Athom's standards 🤞