DiedB/Homey-SolarPanels #273: Envoy does not report to Homey

Olav1971, 2023-07-07:
Since the last upgrade of Zonnepanelen my Homey is not able to fetch the Envoy data. I no longer am able to monitor my production.
“An error occurred while fetching inverter data.” 
reboot of homey did not resolve the issue. The inverter is sending data to Enphase cloud.

RoadXY, 2023-07-07:
I don't know if this is the issue but this week several Envoy's received an update.
At the tweakers.net forum they are still figuring out what has changed.
https://gathering.tweakers.net/forum/list_messages/2075922/52

By the looks of it, you need an installer account to get the info you used to get with a user account...

Olav1971, 2023-07-07:
Thanks, this Envoy update seems the trigger. Still I hope this can be resolved by an update of the Homey app connecting to it.

DiedB, 2023-07-07:
@Olav1971 are you using the latest test version of the app (https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/)? And are you using an Enphase Envoy v7 device (requires removing and re-adding your Envoy to Homey)?

I've implemented it according to Enphase documentation, would be weird if they broke their own rules...

Olav1971, 2023-07-07:
Hello DiedB,
I am using an Envoy with Firmware D7.6.175 .
on the Homey the regular app V5.2.0 .

I had tried removing the Envoy from Homey and added it again. No change in the error message.

Olav1971, 2023-07-07:
Ok, I have installed the experimental zonnepanelen app on my Homey. Removed the  device. Added a new device and selecting the Envoy v7 ( this is a new option?). After completing the credentials for Enphase I received the telemetry again.

ehofstede, 2023-07-08:
> By the looks of it, you need an installer account to get the info you used to get with a user account...

When I tried to access the Envoy locally it wanted me to paste a token. Through https://entrez.enphaseenergy.com/ a token could be requested... That worked for me (not really a straight forward process though) and after entering the token I was able to see the realtime production again. (as a "user" and not as an "installer")

ehofstede, 2023-07-08:
> @Olav1971 are you using the latest test version of the app (https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test/)? And are you using an Enphase Envoy v7 device (requires removing and re-adding your Envoy to Homey)?

I'm using the 5.2.0 version, since it's broken anyway, how can I install and "try" 5.2.2 ?
As told in a reply to RoadXY, when I connected to the Envoy locally it wanted me to enter a token before I was able to see live data again. Could that be the problem here? That the app on Homey doesn't have/use this token..?

I'm not entirely sure if I have a v7 Enphase device, I do think so, the Microinverters are IQ7's, but I guess that's not important, is it? It shows me a serial number, a part number, software version, but not "Enphase Envoy vX"...

ManuelR74, 2023-07-08:
I am quite a beginner with homey pro and currently I am blocking on the same issue. 
The solar panel app got the data from my envoy until yesterday 07.07. morning. Since then, I only get an error message _An unknown error occurred while fetching inverter data_ Quite frustrating error message as it gives you no clue where to search the solution...
Research on several forums seem to confirm that version D7.6.175 of envoy seems indeed to have an issue. Did not get the information wheteher Enphase is aware of it...

ManuelR74, 2023-07-08:
> Ok, I have installed the experimental zonnepanelen app on my Homey. Removed the device. Added a new device and selecting the Envoy v7 ( this is a new option?). After completing the credentials for Enphase I received the telemetry again.

How to install the experimental app ?

RoadXY, 2023-07-08:
> I am quite a beginner with homey pro and currently I am blocking on the same issue. The solar panel app got the data from my envoy until yesterday 07.07. morning. Since then, I only get an error [...]

Might be because of the firmware update.
Did you reboot the Envoy?

ManuelR74, 2023-07-08:
Yes, several times, as well as the homey pro. Also removed and created the device again, as well as unistalling/re-installing the app. Did not improve anything.

RoadXY, 2023-07-08:
Weird, worked first try.
Try and remove the device and add it several days later?
I know Homey remembers the device even if the app is removed...

powercrack, 2023-07-09:
Hello DiedB,
I am using an Envoy with Firmware D7.6.114     
Micro.inverters  firmware 549-00049-r01-v6.05.03
on the Homey the regular app V5.2.3 . V7 app.

I had tried removing the Envoy from Homey and added it again. No change in the error message.

It all stays on  zero

DiedB, 2023-07-09:
> > Ok, I have installed the experimental zonnepanelen app on my Homey. Removed the device. Added a new device and selecting the Envoy v7 ( this is a new option?). After completing the credentials for Enphase I received the telemetry again.
> 
> How to install the experimental app ?

https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test

DiedB, 2023-07-09:
> Hello DiedB, I am using an Envoy with Firmware D7.6.114 Micro.inverters firmware 549-00049-r01-v6.05.03 on the Homey the regular app V5.2.3 . V7 app.
> 
> I had tried removing the Envoy from Homey and added it again. No change . Its make contact. But thats it.
> 
> It all stays on zero

No errors, but values stay on zero?

Could you login to the Envoy by directly going to its IP address, logging in and then going to `https://ENVOY_IP/api/v1/production`? Replace `ENVOY_IP` with the IP address your Envoy is on (sometimes, `envoy` also works). Afterwards, paste the output you get here on GitHub.

powercrack, 2023-07-09:
[image]

powercrack, 2023-07-09:
[image]

powercrack, 2023-07-09:
I have not a digital meter. Next week i have one . Is that the reason?

DiedB, 2023-07-09:
And if you go to `/api/v1/production` after authenticating?

The issue you are facing might be caused by meters being installed but not configured. Maybe you need to click that error and indicate which meter is for consumption and which meter is for production? After that, it will likely start working. Make screenshots if you are in doubt and share them here.

This does not have anything to do with a digital meter, I believe (that is completely separate from Enphase).

ehofstede, 2023-07-09:
I installed the experimental app and add the Enphase, now I do see
measurements again, that's great!
But I don't think they're realtime measurements, is that correct?

powercrack, 2023-07-09:
[image]

powercrack, 2023-07-09:
Can not click the meter to change it in enphase .

powercrack, 2023-07-09:
Almost good . Wish i could help better . But i am limited.  Sorry

DiedB, 2023-07-09:
@powercrack Seems that the Envoy is configured to user current transformers (CT clamps) but they are not installed.

powercrack, 2023-07-09:
Reinstall the app . Nothing changed. Sorry

jcf1975, 2023-07-09:
I have the same problem.
Installed the beta app, but no data from the meters.

ManuelR74, 2023-07-09:
Installed the test (beta) app, and it seems to work for now… Interesting: now I don‘t connect envoy anymore but IQ Gateway during the pairing process.
And also, the data seems to not be real time anymore as it was previously. Not a big issue, but I just noticed it.
I‘ll continue to monitor.

powercrack, 2023-07-09:
> @powercrack Seems that the Envoy is configured to user current transformers (CT clamps) but they are not installed.

Can i do something? I am a user not the installer.  I can do a lot . But i thinking a bit tricky

DiedB, 2023-07-09:
@ManuelR74 what is the delay? Do you have a metered envoy (do you see grid and home consumption)? Which of the metrics seems to be 'live'?

I presume that the latter two are live and frequently updated, while production lags behind?

powercrack, 2023-07-09:
[image]
Dat is wat ik zie aan de hand wat jij mij gestuurd heb .

DiedB, 2023-07-10:
Anyone else getting 0 values, but no errors, like @powercrack does?

I presume the issue is that an Envoy S Metered is installed, while CT clamps are not attached.

ManuelR74, 2023-07-10:
> @ManuelR74 what is the delay? Do you have a metered envoy (do you see grid and home consumption)? Which of the metrics seems to be 'live'?
> 
> I presume that the latter two are live and frequently updated, while production lags behind?

Yes, indeed approximately 15 minutes. When hovering over the icon, I can see it. Sorry, as I said, I'm a beginner
[image]

Interresting that the previous version of the app (the stable one, the one that does not work anymore in my case) was 'live'

powercrack, 2023-07-10:
> Anyone else getting 0 values, but no errors, like @powercrack does?
> 
> I presume the issue is that an Envoy S Metered is installed, while CT clamps are not attached.

I checked the whole installation and yes they have not install the ct clamps! I was not aware the special clamps must be installed.  But now i do . Not my fault but the mechanic  must do. 

So all the credits for you Diedb  for all your spare time . 
I learned a lot of it and hopefully otters.

So now i now what to do. 
Again  , thank you verry much!

jcf1975, 2023-07-10:
But also with the clamps installed its not working.
I dont get any data in the homey app.

powercrack, 2023-07-10:
Maybe not installed the meters?

jcf1975, 2023-07-10:
Just delete the app and installed it again.
Stil no data.

powercrack, 2023-07-10:
Are the meters good install ?

jcf1975, 2023-07-10:
Where can i activate the readers?

powercrack, 2023-07-10:
Aaah okay . It was okay in the past in the homey app. 
I did know . Sorry. Then it is activate.  So far i know
Maybe restart de panels but, maybe its wrong . I am new here . I read a lot over it

powercrack, 2023-07-10:
Maybe a pro can help you.

DiedB, 2023-07-11:
@jcf1975 The new app and the new Enphase API uses the CT readings (clamps) if your Envoy is Metered. The app seems to be working fine, but the values indicate that the clamps are not installed correctly.

DiedB, 2023-07-11:
@ManuelR74 The update frequency for production has been fixed in the latest version.

jcf1975, 2023-07-11:
This systeem is not metered with clamps. I recieve the information from the system (micro inverters) .

powercrack, 2023-07-11:
[image]
No ct clamps installed but now i have more working here

jcf1975, 2023-07-11:
> [image] No ct clamps installed but now i have more working here

But there is still no "Energie opwek vandaag" and that information is availeble from the micro inverters.

powercrack, 2023-07-11:
So ct clamps are not necessary?

DiedB, 2023-07-11:
Did some more research on this.

When you have an Envoy S Metered, it normally comes with a set of CT clamps. If the installer enables them in Envoy software and installs them correctly, the Envoy will report production (and if set up, consumption) data and it will show up in this app. This applies to most users.

@powercrack and @jcf1975, you have an Envoy S Metered, where the installer enabled CT clamps in the software, but did not install them (or at least not correctly). The Envoy will report data from the CT clamps, but because they are not installed, this is bogus data.

You have two options:
- Correctly install and set up the CT clamps that came with the Envoy
- Disable CT clamps in Envoy software (if disabled, the Envoy will report data from the inverters to Homey and the app will work correctly)

For both options, you will likely need to contact your installer.

closed 2023-07-11 by DiedB

powercrack, 2023-07-11:
> Maybe one more thing, what do you get when you go to http://envoy/api/v1/production/inverters?

I can , but give me 10 min . And i post it

jcf1975, 2023-07-11:
Thx for the info. Clamps will never come with a meter in holland. Its an option that almost nobody buy. I contact this afternoon the support from enphase and this is the information the give me.

powercrack, 2023-07-11:
[image]

powercrack, 2023-07-11:
I think i have some problems  because off   the weather i think.
Let me try again.

powercrack, 2023-07-11:
At the moment not possible , sorry about that . 
Tommorrow morning i will try again.

jcf1975, 2023-07-11:
This is my info...
[image]

powercrack, 2023-07-11:
[image]

jcf1975, 2023-07-11:
> [image]

Youre meters are also disabled.

powercrack, 2023-07-11:
[image]

powercrack, 2023-07-11:
> > [image]
> 
> Youre meters are also disabled.

Yes ,i see it too. Thanks

powercrack, 2023-07-12:
[image]

powercrack, 2023-07-12:
[image]