DiedB/Homey-SolarPanels #275: Envoy heeft update en nu werkt de zonnepanelen app niet meer.

spikehome, 2023-07-08:
Vanmorgen gingen mijn screens allemaal niet naar beneden na even gekeken te hebben zag ik dat de envoy een error gaf.
Device verwijderd en opnieuw toegevoegd, maar precies dezelfde melding.
Toen op IP naar de envoy met de browser en kreeg ik de melding dat het nu https is en ik moet inloggen met het enphase account.

Hoe is dit nu weer werkend te krijgen?
Op de envoy kan ik niks instellen zover ik kan zien.

RoadXY, 2023-07-08:
Je hebt waarschijnlijk een update gehad van versie 5 naar versie 7.
In de test versie is er ondersteuning voor ingebouwd

spikehome, 2023-07-08:
Ok tnx, test versie geinstalleerd v7 die werkt, dat is zo te zien ook live power output.
tnx

RoadXY, 2023-07-08:
Graag gedaan. Gaan er wel meer komen, ben ik bang

DiedB, 2023-07-09:
Ik bouw een melding in als een v7 Envoy wordt gedetecteerd terwijl de oude driver wordt gebruikt.

closed 2023-07-09 by DiedB

spikehome, 2023-07-11:
Wat jammer is dat Enpase dit soort dingen zo maar aanpast.
Ik ga een shelly pro 3 er tussen zetten waarmee ik de terug levering meet.
Ik schakel namelijk zeer veel op het vermogen wat terug geleverd word zonnewering b.v.
Tnx iig het werkt nu allemaal weer wel even alle flows na moeten lopen.