DiedB/Homey-SolarPanels #276: Power production in insights

ManuelR74, 2023-07-11:
I use the beta version of the Solar Panel app for Homey pro. 
Currently, the historical power production data is not visible in insights. I would personally very much appreciate this feature. 
Is there any plan to implement it ?

[image]

DiedB, 2023-08-19:
That is very weird. Have you tried re-adding the device to Homey?

ManuelR74, 2023-08-20:
I think the translation is wrong (my phone and therefore my app is in german). Looking closely at the figures, I think the label „Energieverbrauch“ (=energy consumption) should actually be „Power production“