DiedB/Homey-SolarPanels #278: Error during first install

Frank-Rutjens, 2023-07-11:
Hello,

during First install I receive the following message:
System ID or API key are incorrect, please check your settings.
After that I can only Select OK, but that does not get me forward.
My only option is to abort.

I ran an diagnostic report:
3434fb05-277e-42aa-9065-495ffe53466e

I'm trying tot install an solar edge SE 4000H 1 fase HD Wave.

What am I doing wrong?

Regards Frank

DiedB, 2023-08-19:
Hi Frank! This often happens when you get the API key off of SolarEdge's website, without clicking 'save'. Could that be the issue?

Frank-Rutjens, 2023-08-21:
Hello Diederik,

I contacted the installer of the solar panels.
He send me a new api-key.
With it I was able to install de inverter, but I get the error message as
can be seen in the attached image.

I hope you are able to assist me.
Thanks in advance.

With kind regards
Frank.