DiedB/Homey-SolarPanels #279: Solis Cloud / Ginlong not working anymore with the last update

mvkooten1984, 2023-07-11:
Solis Cloud / Ginlong not working anymore with the last update
is there any reason for that the not working anymore ? or become there a next update with support for this item

mvkooten1984, 2023-08-28:
is there any update when it is coming back in the app ?

Bloke63, 2023-08-31:
Volgens chatGPT:

Ik heb wat onderzoek gedaan naar uw vraag en ik heb het volgende gevonden:

- SolisCloud is een portaal voor Solis PV-systemen die een API biedt om gegevens te lezen van de SolisCloud-service¹².
- Homey is een slimme thuiscontroller die verschillende apparaten en diensten kan integreren via apps⁴.
- Er was een app genaamd Solar Panels die SolisCloud kon integreren met Homey via de Ginlong API³, maar deze app is niet meer beschikbaar in de Homey App Store⁵.
- De reden waarom de app is verwijderd, is omdat de Ginlong API is stopgezet in mei 2023 en vervangen door de SolisCloud API. De ontwikkelaar van de app heeft geen plannen om de app bij te werken om de nieuwe API te ondersteunen.
- Er is momenteel geen andere app die SolisCloud kan integreren met Homey, voor zover ik weet. U kunt proberen contact op te nemen met de ontwikkelaar van de Solar Panels app of een verzoek indienen bij de Homey community om te zien of iemand bereid is om een nieuwe app te maken.

Ik hoop dat dit antwoord u helpt. 😊

Bron: Gesprek met Bing, 31-8-2023
(1) SolisCloud sensor integration - GitHub: Let’s build from here. https://github.com/hultenvp/solis-sensor.
(2) Soliscloud gebruikershandleiding voor systeemeigenaars en/of .... https://solis-service.solisinverters.com/nl/support/solutions/articles/44002251720-soliscloud-gebruikershandleiding-voor-systeemeigenaars-en-of-organisaties-installateurs-.
(3) undefined. https://www.soliscloud.com/.
(4) Ginlong/Solis support in Homey v2.0 #9 - GitHub. https://github.com/DiedB/Homey-SolarPanels/issues/9.
(5) undefined. http://m.ginlong.com/main.html.

FiDoen, 2023-10-01:
Please add support for Solis Cloud @DiedB 😇

FiDoen, 2024-01-14:
DiedB will probably never make a Solis Cloud compatible app.
Recommends everyone who has Solis Cloud to donate / use new Solis Cloud Inverter app:

https://homey.app/en-no/app/com.soliscloud.inverter/Solis-Cloud-Inverter/