DiedB/Homey-SolarPanels #289: Zonnepanelen verliest contract met SolarEdge

Nakika, 2023-07-25:
[image]

het ging een tijd goed. Ik laat de app nu weer ieder uur herstarten. Dan werkt het weer. 

47878154-0756-422b-8a92-272d4db6a46e

DiedB, 2023-08-19:
Heb je dit issue nog steeds? Zo ja, zou je je API key met mij kunnen delen via slower_04_machete@icloud.com? Dan duik ik er even in.

Nakika, 2023-08-19:
> via ***@***.***? Dan duik ik er even in.

Hi Diederik,

Het euvel is vanzelf verdwenen zonder update van jouw kant. Ik begin te
vermoeden dat het probleem bij SolarEdge lag.

Hartelijk dank voor je reactie en nog een fijn weekend verder.

Groet Ria Banis

closed 2023-08-19 by DiedB