DiedB/Homey-SolarPanels #291: Growatt SP4000H with Batterie not supportet

chrispeterpdm, 2023-08-09:
I have 3 Inverters from Growatt. 2 Inverters from Growatt 2000-S and MIC 3000TL-X are working. But the Inverter SPH4000H can not be found. But this is my important Inverter, because he have not only the Powerinformation, he have also the Information from the Batterie. Is it possible to support this Inverter? Greetings Chris

DiedB, 2023-08-19:
Do you have information about the SPH4000H in the ShinePhone app?

chrispeterpdm, 2023-08-19:
Hello, what Kind of information you need?Greetings Chris --Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

chrispeterpdm, 2023-08-19:
Here some hardcopies--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

DiedB, 2023-08-19:
I was wondering whether you are able to access the data of the inverter and battery in Growatt's app, to determine whether the SPH-series is connected through their API.

chrispeterpdm, 2023-08-19:
I can access the data, but I can't find anything about the API--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

chrispeterpdm, 2023-08-19:
I know, however, that the Clever-PV portal retrieves the data correctly via the API--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

chrispeterpdm, 2023-08-19:
Here a hardcopy from clever-pv...--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.