DiedB/Homey-SolarPanels #293: Growatt integration is not working anymore

dhavyd, 2023-08-16:
Hi! Thanks for a great app that has helped me a lot the last year... I have a Growatt MOD13KTL3-X and has worked without any bigger issues. Yesterday it lost the connection to the API. I've tried to connected again. I've also removed the device from Homey and added again, but I still get the message: "Could not login to Growatt server".

It works to login through the Growatt ShinePhone app. But, on ShinePhone, you need to selected "I agree to..." every single time you log in. This is new too that appeared the lastest days. I suspect that is the issue. I seems like Growatt now requires you to "agree" to their terms everytime you login. Maybe it's why you can't login to the API with only username and password.

erlendoyen, 2023-08-16:
Same her.

Growatt tibber integration is still working, but they have some agreement for common api access.

pro-sumer, 2023-08-16:
Also no data since yesterday.

At first sight it looks like the “log in” request itself gets a success response, but all requests after that fail because I’m not logged in?

(then this might be a server issue that Growatt needs to fix?)

dhavyd, 2023-08-16:
But if you remove the device then it's not possible to add it again, since the login fails...

pro-sumer, 2023-08-16:
> At first sight it looks like the “log in” request itself gets a success response, but all requests after that fail because I’m not logged in?

That was with [HTTPBot](https://www.httpbot.io). API still seems fine in [RapidAPI for Mac](https://mac-docs.rapidapi.com).

Ivarrrrr, 2023-08-18:
Experiencing the same issue. No way to log in anymore.

Didiervu91, 2023-08-18:
Yup, login failed here too.

dyandh, 2023-08-19:
Hi, experiencing the same issue: existing devices not receiving data an unable to add new devices. Reinstalling the app is not helping, login using the Shinephone application is working, but as stated, now needing to select "I agree to...".
Changes on the Homey: upgraded 3 days ago to Homey v 10.0.0. Not sure if this is related

DiedB, 2023-08-19:
Debugged this for quite some time but I can't get a grip on the issue. The app is hitting some Alibaba Cloud WAF. I get the same response doing the request through Postman. However, making the request through curl on my machine seems fine. Proxying the request from the Homey app through my machine also works fine.

Not sure what property on the request from Homey is deviating from the behaviour of the setups that do work. Order of request headers, content of request headers and source IP do not seem to be the issue here. Any ideas? Maybe something in the TLS setup?

[image]
Postman (failing)

[image]
curl (working)

[image]
Homey (failing)

[image]
ShinePhone (working)

pro-sumer, 2023-08-19:
I don't understand either.

In addition to HTTPBot and RapidAPI I now tried the GUI and CLI version of [HTTPie](https://httpie.io/):
* CLI: works
* GUI: fails with error 405 and explanation "_Sorry, your request has been blocked as it may cause potential threats to the server's security._"

pro-sumer, 2023-08-19:
> The app is hitting some Alibaba Cloud WAF.

Yes, noticing the same:

```
➜ nslookup server-api.growatt.com
Server:		1.1.1.1
Address:	1.1.1.1#53

Non-authoritative answer:
server-api.growatt.com	canonical name = z93rc2bjmoqbszjwwodwz1hcgn3pfsha.aliyunwaf2.com.
Name:	z93rc2bjmoqbszjwwodwz1hcgn3pfsha.aliyunwaf2.com
Address: 47.254.137.157
```

Roylegr, 2023-08-19:
Same here can't login anymore. Homey Pro 2019. Growatt

Grom7206, 2023-08-20:
Same here. Can't login anymore. Homey Pro 2019. Growatt.
Error: Could not login to Growatt server

pro-sumer, 2023-08-20:
@DiedB Not sure whether this will help, but other tools accessing the Growatt API run into the 405 error as well.

[One suggestion](https://github.com/sincze/Domoticz-Growatt-Webserver-Plugin/issues/40#issuecomment-1685203267) is to use `server.growatt.com` instead of `server-api.growatt.com`. When I try this in the HTTPie GUI the request still fails, but now because the SSL certificate does not match that domain. If I ignore this and send an insecure request, the login request indeed does "succeed".

DiedB, 2023-08-20:
@pro-sumer Thanks, that seems to work well indeed. Let's try that and hope it keeps working for now. I have submitted a new build with Athom, you can also already use the test version (https://homey.app/nl-nl/app/it.diederik.solar/Zonnepanelen/test).

closed 2023-08-20 by DiedB

dyandh, 2023-08-22:
Thanks all, it is working again as before 👍

dhavyd, 2023-08-22:
Yes, it's working again, thanks!

erlendoyen, 2023-08-23:
Working again for now. 

Shinephone app changed again, agreement checkbox removed from signing box, popup after signing.

Also scheduled maintenance

[image]

Wouwmw, 2023-08-23:
Works here too. Thanks for your quick support