DiedB/Homey-SolarPanels #295: Growatt SPH3600 supported?

ridderdk, 2023-08-30:
Growatt SPH3600 not yet supported?
When adding unit Solar panels - Growatt and logging in, it returns "no devices found"
I can see my device in ShinePhone app, and from Home Assistant, i can read all details

DiedB, 2023-12-23:
I can have a look, but need to check with your ShinePhone credentials to see if this could work. If you are open to share them with me, you can send an email to [slower_04_machete@icloud.com](mailto:slower_04_machete@icloud.com).

closed 2023-12-23 by DiedB

llBIKSll, 2024-05-05:
> I can have a look, but need to check with your ShinePhone credentials to see if this could work. If you are open to share them with me, you can send an email to [slower_04_machete@icloud.com](mailto:slower_04_machete@icloud.com).

Hi DiedB, was is solved? I am asking as I have the same problem with my SPH3600. "No New Device found"