DiedB/Homey-SolarPanels #299: Enlighting user id en api key

evinxnl, 2023-09-15:
na het installeren van de app in homey wil ik mijn zonnepanelen toevoegen

hiervoor heb ik een user id en api key nodig
die user id heb ik gevonden maar die api key kan ik dus nergens vinden wat doe ik niet goed ?

graag oplossing ?

RoHu71, 2023-09-20:
Ik zou dat graag ook willen weten. API client, Client secret ect alles al geprobeerd maar deze werken niet.

evinxnl, 2023-09-29:
waarschijnlijk werkt het niet omdat de code nogsteeds gebruik maak van v2 api en die wordt niet meer ondersteund ze zijn overgegaan naar api v4

https://developer-v4.enphase.com/docs.html

update de app aub !!

DiedB, 2023-09-30:
Je issue is een duplicate, zie #235. 

Ik raad je aan de andere Enphase-integratie te gebruiken, als dat mogelijk is.

closed 2023-09-30 by DiedB