DiedB/Homey-SolarPanels #301: New User - Unable to find option With Envoy - No New Devices have been found with option Enpahse IQ Gateway.

Onedutch, 2023-09-21:
Hello,

Fresh 2019 Homey Pro, added the "Zonnepanelen" app, I found a thread [https://community.homey.app/t/enphase-enlighten-zonnepanelen-koppelen-aan-homey/75239/3](url)  which says , choose the "via envoy" option. I only have "Enphase Enlighten" and "Enphase IQ Gateway" . I'v read about issues for the API optione V2 and or V4, so I skip the "Enphase Enlighten", because it asks for a API key. 
When choosing the "Enphase IQ Gateway" I first thought I need to use the Envoy Local username:password. I tried the envoy:XXXX (I changed the default password, last 6 digits from the serial) combination, no luck, i tried the installer:XXXXX combination also no luck. Then I tried the email:XXXXX combination I use to logon the [https://enlighten.enphaseenergy.com/](url)  and yes, it starts. Withing 5< seconds it reports "No new devices have been found." I never used Homey, and I never have been connecting my Enlighten account to another Homey or something. 

Unsure what I'm missing ;)

Best Regards,
Onedutch

DiedB, 2023-09-24:
Could you try navigating to your Envoy (IQ Gateway) in a browser and authenticating there? You might be able to find its IP address in your router or in the Enphase app.

Let me know what version the IQ Gateway is running on (/info.xml).

Onedutch, 2023-09-29:
Hello DieB, 

It's : 
Envoy
Serial Number
1220xxxxxxxxx
Part Number
800-00554-r03
Software Version
R4.10.35 (6ed292)
Software Build Date
13 Nov, 2018 2:57 AM
Wi-Fi Mac Address
2C:xx:xx:xx:xx:xx

All the described attempts are on the Homey App, I tried to autenticate (with the installer login) on my browser first to check if the username/pass are OK. 

What is curious, so on the browser with the local IP I can press Installer Login at the bottem and user installer as username and the password, this works. It does not while adding the device in Homey. 

I'm unable to test the 'normal' login, reading on the internet the username should be envoy and using a password. I changed the password again, (first trid an fault password, it say, 'old password' not correct) to check if the envoy password is ok. using envoy as username and this new password still say .

So to recap, when I choose Homey / Solar Panels / Enphase IQ Gateway, I should use the crdeentials I use to login the local Enpahse IQ Gateway.

Thanks for your time!
OneDutch

DiedB, 2023-09-30:
Hey, thanks for the data! Your Envoy runs on a pretty old software version. Does it have access to the internet?

https://support.enphase.com/s/question/0D53m00006f6b72CAA/firmware-update

You need to log in to the IQ Gateway itself and update the firmware there (it should be R7.x for it to work with Homey). Is there an option for that by logging in as installer?

Onedutch, 2023-09-30:
Yup, seems that the IQ Gateway isn't IQ enough to update itself to something new.. I Logged in with installer login, unable to find something which sounds likeupdate. Yes it is connected to the internet allright.. It even sends (or pull) unsure how it works, everything to https://enlighten.enphaseenergy.com/ where I can see all statistics. 

Now I recall, I think the company I bought the equipment from had to 'activate' de IQ Gateway. I think I have two options to update de IQ Gateway:
- Contact that company, they can search my IQ Gateway in their Enlighten Manager software  and press Update Site or something
- Download an Adroid App and Update myself [https://support.enphase.com/s/article/how-to-update-the-software-of-an-iq-gateway-using-the-enphase-installer-app](url)

Any drawbacks with this never Firmware ?

This will take some time, I'm not @home 

I keep you posted on my progress.
Ciau , Onedutch

DiedB, 2023-09-30:
One drawback is that you become dependent on Enphase for access to the Envoy, because they use your Enphase account for (local) authentication. But it is good for security and the version you're currently on will become obsolete, so I would just go for the update to R7.

Onedutch, 2023-10-12:
I checked the Android App to update myself, need an account for it. I will wait and see what wil happens, thanx for the support.

closed 2023-10-12 by Onedutch