DiedB/Homey-SolarPanels #303: Geen verbinding met Growatt server

Dennis25101968, 2023-09-28:
Sinds gisteren  krijg ik op de Homey Pro geen verbinding meer met de Growatt server.
In de Growatt shine app en/of op server.growatt krijg ik wel verbinding.
Ik weet niet hoe dit op te lossen is.

martijngeerts, 2023-09-28:
Hier hetzelfde probleem weer. Het ging na de laatste update verrassend goed, tot gister.

pro-sumer, 2023-09-28:
Log in seems to fail again (with 405 or 501 errors).

The Homey App broke this morning (Thursday), my iOS Shortcut on Tuesday.

No workaround this time?

pro-sumer, 2023-09-30:
More API users are noticing this::

https://github.com/indykoning/PyPi_GrowattServer/issues/74#issuecomment-1741335616

Let’s keep an eye on that issue.

DiedB, 2023-09-30:
Thanks for tracking this. Happy to help once a solution is found.

pro-sumer, 2023-09-30:
Is anyone still able to use the ShinePhone App?

I can’t even log in using that official Growatt App...

(so maybe a server is down?)

DiedB, 2023-09-30:
ShinePhone doesn't work for me either.

Finnsen, 2023-09-30:
I am not able to log into the server via the Homey App, but my ShinePhone is working fine.

pro-sumer, 2023-10-01:
ShinePhone is working again for me, but everything else trying to log in still fails.

dyandh, 2023-10-02:
Any update/progress on this issue? Since a week of so I have the same problem: my 2 Growatt converters will not connect. I have removed one from Homey, but I am not able to redefine/add this back in Homey. The error is "could not login to Growatt server". The ShinePhone application is working normal.

Bennie1954, 2023-10-03:
Ik heb het zelfde probleem sinds een kleine week werkt de app voor Homey niet meer. Ik heb twee Growatt omvormers de op Shinephone werkt normaal

dhavyd, 2023-10-06:
Same issue here... I believe this is related to #293 since "Accept agreement" need to be selected in the ShinePhone App in order to login.

pro-sumer, 2023-10-13:
We might benefit from a discovery by @vtraveller!

Using the web API (instead of the mobile App API) I was able to log in and obtain the two values we need.

Need to find time to write a small proof-of-concept with demo code...

pro-sumer, 2023-10-15:
@DiedB Can you please try the Growatt web API instead of the mobile API?

Example code:
```
import axios from 'axios'

const client = axios.create({ 
  baseURL: 'https://server.growatt.com', 
  headers: { 'Content-Type': 'application/x-www-form-urlencoded; charset=utf-8'},
})

async function testWebAPI () {
  // Log in
  const loginResponse = await client.post('login', 
    { account: 'USERNAME', password: 'PASSWORD' }
  )
  const setCookie = loginResponse.headers['set-cookie']
  const plantId = setCookie.filter(cookie => cookie.includes('onePlantId'))[0].split(';')[0].split('=')[1]
  const cookie = setCookie.map(cookie => cookie.split(';')[0]).join(';')

  // Get production data
  const devicesResponse = await client.post('panel/getDevicesByPlantList',
    { currPage: 1, plantId },
    { headers: { Cookie: cookie} }
  )
  const data = devicesResponse.data.obj.datas[0]
  console.log(`Energy today: ${data.eToday}`)
  console.log(`Current power: ${data.pac}`)
}

testWebAPI()
```

Notes:
* Only works for 1 plant (OK for most home users?)
* Hardcoded for 1 inverter (but you can iterate over `datas` instead of only using element 0?)
* Requires (unmodified) website credentials (no "hashing" of password as used by the mobile API)
* Results are strings in a certain locale (using a dot as decimals separator?)

bkruyver, 2023-10-17:
Same issue here. Would be nice to have this working again. Is there no way to read data directly from inverters?

pro-sumer, 2023-10-23:
> Is anyone working on this?

I was hoping that I could test [using the web API](https://github.com/DiedB/Homey-SolarPanels/issues/303#issuecomment-1763479698) by changing a few lines in `api.ts` (and hardcoding some parameters for the experiment), but the source code has changed quite a bit since I wrote the first implementation of the Growatt driver. This means that I need more time to do such an experiment, which I don't have right now.

Therefore, I hope that Diederik (or someone else capable) has the time to work on this. If not, I might pick this up again at a later point.

(First results from some experiments with migrating from the mobile API to the web API in other personal JavaScript projects that use the Growatt API are promising)

mijdrecht, 2023-10-26:
After creating a new  (Sub) account on the growatt website (under settings) , I was able to use the app with the new account.

dhavyd, 2023-10-26:
I did the same now and can confirm that the app works if you create a sub account, thanks @mijdrecht !

pimviergever, 2023-10-26:
> After creating a new  (Sub) account on the growatt website (under settings) , I was able to use the app with the new account.

This worked for me, also. Thank you!

martijngeerts, 2023-10-26:
> After creating a new (Sub) account on the growatt website (under settings) , I was able to use the app with the new account.

This worked for me as well :) thx!

Limbeckx, 2023-10-26:
Very strange... When I do the same, I get the new Devices Found screen with the undertext: Select devices you would like to add. The list is empty for a while and afterwards I get the error saying 'Timed out after 30000ms'

Kammiz, 2023-10-26:
> After creating a new (Sub) account on the growatt website (under settings) , I was able to use the app with the new account.

Awesome, this works!

pro-sumer, 2023-10-26:
> the app works if you create a sub account

Nice find!

PS: SolarPanels did not accept the credentials of the new sub account until I restarted it

bkruyver, 2023-10-26:
Could someone please give a howto for making a new account? I can not find a way to do it.....
Thx

mijdrecht, 2023-10-26:
Logon to the growatt website with your current account.
Goto settings
Create a new account.

bkruyver, 2023-10-26:
Thx, i found out that i can only see the installer account. I will create my own private user account this way and see what happens

pro-sumer, 2023-10-28:
> PS: SolarPanels did not accept the credentials of the new sub account until I restarted it

In fact, my new credentials were not stored before the restart...

SolarPanels apparently is working again with my regular (main) account!

dyandh, 2023-11-04:
Hi, same experience: since a couple of days ago SolarPanel is able to connect again with my original settings/account. Not sure if changes were made, if so thanks for the support.
Dyan

MP-ICT-Diensten, 2023-11-17:
November 2, 2023 at 13:00 I have the first connectivity again, all is working fine again.
Connectivity was down from September 28, 2023 I can see in Insights.
Growatt has changed/updated/fixed something probably.

closed 2023-12-23 by DiedB