DiedB/Homey-SolarPanels #309: Add support for SUNGROW inverters

seppil, 2023-10-23:
Is support for sun grow inverters planned in the (near) future ?
I have a Sungrow SG3K6-D and though I have not the programming skills I am very much willing in helping out in any way to get the Sungrow inverters going.

hAMMERlized, 2024-01-12:
Would appreciate it if you could add support for sungrow inverter. I have implemented mine in home assistant via modbus so far.

Any idea when of if it will be supported? Thanks.

burrikido, 2024-02-05:
+1 for SunGrow, thank you

Callbeloved, 2024-02-17:
Appreciate if you can add sungrow Inverter please ? 

I have Sungrow inverter model SG10RT

Also happy to help in anyway possible in contributing towards coding (not an expert but a beginner to mid-level experience. 

Thanks, Dave