DiedB/Homey-SolarPanels #316: geen dag opbrengst en alles is - negatief

evinxnl, 2023-11-25:
alles is negatief en er is geen dag opbrengst
[image]

maar zoals je kan zien enphase app laat wel degelijk iets zien dat er opgewekt is vandaag ?
[image]

ik hoop dat je dit kan oplossen de app is een groot aanwinst voor dashboard weergave aan de muur 😃

keep up the good work 😃

HPP1234, 2023-12-05:
Ik heb precies hetzelfde probleem met de Enphase IQ Gateway.
Geen idee hoe dit is op te lossen. Graag wel een oplossing als dit mogelijk is..
Harry Postma

DiedB, 2023-12-23:
Je hebt een Metered IQ Gateway, welke door de installateur aangesloten hoort te worden in je meterkast met current clamps. Dit lijkt hier niet correct geinstalleerd te zijn.

De app van Enphase heeft de data wel, omdat ze de som nemen van alle micro-omvormers op het dak (deze rapporteren ook). Echter geven ze in de API van het apparaat de data van current clamps door, en Homey is hiervan afhankelijk. Ik raad je aan contact op te nemen met de installateur om dit goed te installeren.

evinxnl, 2024-02-07:
beste bedankt voor je bericht na dit bericht heb ik contact gehad met leverancier van de panelen en zelf de gateway opengemaakt en er waren inderdaad geen CT sensoren gemonteerd vandaag is de monteur geweest om deze te installeren 

alles werkt nu naar behoren super bedankt voor je bericht en uitleg

[image]

evanwijland, 2024-08-28:
Ik wil graag inhaken op deze post:

Bij mij heeft alles tot nu toe goed gewerkt, alleen sinds eergsiteren kreeg ik een verbindingserror en na het herinstalleren van de Zonnepanelen app en opnieuw verbinden met de Envoy krijg ik ineens negatieve waarden.

Ik heb geen clamp verbonden (deze paste niet), maar het uitlezen heeft altijd naar behoren gewerkt. 

Kan het zijn dat de Envoy update er voor heeft gezorgt dat de API nu alleen maar de waarden van de clamps uitleest? Op de Envoy portal is wel alles netjes uit te lezen.