DiedB/Homey-SolarPanels #321: Koppelen Enphase aan Homey

RNLMC, 2023-12-21:
Het lukt mij niet mijn enphase te koppelen aan homey. Lees dat d.m.v API geen optie is en via de IQ gateway lukt het niet. geeft aan na invoeren ww dat er geen nieuwe apparaten zijn toegevoegd. Iemand een oplossing?

DiedB, 2023-12-23:
Hoe ziet je netwerk eruit, zit Homey in hetzelfde netwerksegment als je IQ Gateway? Welke versie van de IQ Gateway heb je?

RNLMC, 2023-12-27:
Hoi, mijn Homey is met WiFi verbonden en mijn IQ middels een kabel, bij firmware staat D7.6.175
[image]
[image]

DiedB, 2024-01-04:
Vermoedelijk is dit een netwerk-probleem. Zou je je Enphase Envoy (ook) met WiFi kunnen verbinden? Klik op 'WiFi configureren' op de plek waar je de screenshot maakte en volg de stappen om de Enphase aan hetzelfde WiFi-netwerk te koppelen als je Homey. Vervolgens kun je proberen opnieuw met de Enphase te verbinden.

Een alternatief is om een Ethernet-adapter aan te schaffen voor de Homey (https://homey.app/nl-nl/store/product/homey-pro-ethernet-adapter/) en deze bedraad aan je netwerk te verbinden.

Een laatste alternatief, maar technisch complex, is het repeaten van mDNS tussen je LAN en je WiFi-subnet. Maar ik zou een van de eerste twee opties overwegen als je niet heel technisch onderlegd bent.

MacManiac78, 2024-06-26:
Kan in het geheel niets van Enphase aan Homey toevoegen.

vierlijner, 2024-08-02:
#328