DiedB/Homey-SolarPanels #324: Partial Date Enphase in Homey

fransjvos, 2024-01-18:
Hello Diederik,
Today new solar panels with Enphase were installed. I created a new device in my Homey, however I do not see all the data.
As you can see in the screenshot I can see the "Energie Opwek vandaag" and the value is updated regularly. The other values "Vermogen" remain Zero or empty. What can I do to resolve?
Thanks in advance,

Frans Vos

[image]
[image]

fransjvos, 2024-01-18:
Searching trough Facebook I understand there are 2 ways of connecting: directly trough Envoy or trough Enlighten (cloud?). I only seem to find the latter, which seems to work in completely. How to connect trough Envoy? I use HP23.
Thanks in advance!

Hooiwagen44, 2024-01-18:
Ik heb hetzelfde probleem maar dan met een Envoy-S-Metered