DiedB/Homey-SolarPanels #328: Cannot connect Homey to Enphase IQ Gateway

emilykgrucza, 2024-03-12:
Hi,

I have tried searching for a solution to this issue but no luck or got confused by the translations. I have downloaded the app onto my Homey Pro 2019 (on the latest firmware) and have enphase solar panels.

Once I add my credentials I get the following error 
[image]

Not sure what details you require but the enphase gateway is on firmware R4.10.35 and is Envoy-s-metered-EU

Solar panels inverters are IQ7+

gateway is connected to my home network via Ethernet not wifi.

I have checked as far as my knowledge has allowed if there were updates for enphase.

currently it says I have no Applications using the API, I received no request to access the api when we set it up.

some guidance would be greatly appreciated. Thankyou Emily

emilykgrucza, 2024-04-20:
> I had the same problem as you, though I was on the R5 firmware, not R4. Apparently, Enphase doesn't do automatic major updates, unless you ask for them (I guess to prevent problems that might occur with major updates). I contacted Enphase for an update to version 7. After they pushed the update to my Envoy gateway, the app worked fine. You'll need your location-id for this.

Thankyou so much for your help. The upgrade has solved the connectivity issueThanksEmily

JrW10, 2024-10-28:
Same problem here. I removed the device, reinstalled the app, but now the device can't be foud anymore?
Homey Pro 2023 version 12.1.2
App Zonnepanelen v5.3.2
Enphase firmware D8.2.4225

JoostBerk, 2024-10-29:
So i don't know exactly what I did but I de-installed the Zonnepanelen app
from Homey, and a few weeks later I reinstalled it and then it was able to
discover the IQ again. Maybe just leave it for a while and then try it
again.