DiedB/Homey-SolarPanels #330: No connevtion to Enphase

Tonlaar, 2024-03-20:
Since March 15 I haven't had a connection with my Enphase system.  
As a last resort today I removed the Zonnepanelen app and tried to reinstall the app. As it turns out the App is loaded anymore. It's not on the list of apps on my Homey. Restarted the Homey but it's still not there.
What do I need to do te reload the Zonnepanelen App on my Homey?

GitMike1233, 2024-03-20:
Very likely your Envoy , whether unsolicited or not, has been updated to version 7.xx.
You can try logging in with your name and password for your Enlighten app. Maybe that will work.

Tonlaar, 2024-03-21:
Thanks, GitMike1233. As stated in my original message I did reinstalled the App, entered my name and password and it all worked again. I was curious about what caused the interruption. It's good to know that an update of the Enphase software took place that may have caused the issue.