DiedB/Homey-SolarPanels #338: Toevoegen Enphase IQ converter wert niet

cbogers81, 2024-06-21:
Hallo,

Ik probeer Enphase IQ converter toe te voegen aan de Homey Pro 2023. Maar na het inloggen krijgt ik de melding: geen nieuws apparaten gevonden.

Zie screenshot als bijlage.

Met vriendelijke groeten,
Chris Bogers
[image]

MacManiac78, 2024-06-26:
Same here, Enphase can't be added on Homey

vierlijner, 2024-07-01:
You need to connect with the gateway Envoy. And they need to be on the same network as the homey. This was my first problem, I had it on the guest network and on my deco I had to disable the rule: in isolation.

cbogers81, 2024-07-01:
> You need to connect with the gateway Envoy. And they need to be on the same network as the homey. This was my first problem, I had it on the guest network and on my deco I had to disable the rule: in isolation.

What do you mean on the same network? My mobile phone is connected to the same wifi as the homey. On my mobile I can access the gateways of Envoy. I'm using the same credentials as in the Enlighten app. So this should work right?

Or are you saying, you must connect your homey via LAN instead of WIFI?

vierlijner, 2024-07-01:
No, you can leave it on WiFi, that should be fine.
The enlighten app is going by the Internet/ cloud and not over the internal network. But the homey will go directly to the device.
The WiFi name (and password) should be the same on the envoy as on the homey.

vierlijner, 2024-07-01:
Can you access it on you mobile browser on https://192.192.192.192/home where 192.192.192.192 should be replaced with the ip Adres of the envoy?

cbogers81, 2024-07-01:
Yes I can.

vierlijner, 2024-07-01:
Okey, than remove device and app from homey and try again.

cbogers81, 2024-07-01:
What do you mean with remove the device?
I don't have any device for the Zonnepanelen app, because nothing is found.

I can only remove the app itself and install it again. Already did this many times.

vierlijner, 2024-07-01:
Than @DiedB has to take over it and you have to maybe send the diagnostic information.

cbogers81, 2024-07-01:
> Than @DiedB has to take over it and you have to maybe send the diagnostic information.

Just did, edit my first post with ID for developer.

cbogers81, 2024-07-06:
@DiedB Is this a know issue? Are you working on this?

MacManiac78, 2024-07-07:
Here Envoy is connected by WiFi and Ethernet. Still the Envoy cant be found with "Zonnepanelen" app in Homey.

JrW10, 2024-10-28:
Same problem here. It used to work just fine. But then Homey lost connection. I removed the device, reinstalled the app, but now the device can't be foud anymore?
Homey Pro 2023 version 12.1.2
App Zonnepanelen v5.3.2
Enphase firmware D8.2.4225