DiedB/Homey-SolarPanels #345: Homey-Pro Solar panels TIGO - NO-GO

koehorst, 2024-11-12:
Hi!  

Perhaps unnecessarily, but my Homey Pro and solar panels do not work with the so-called support for Tigo. My equipment is from Tigo. When I try to add everything, I get the message below (no devices found). Anyone else have problems with this?

Sorry screenshot ( Dutch ) use google lens for translation 

[image]

patupatu, 2024-11-22:
Hi, 
I'm encountering the same error as you.
I have just installed a system with a tigo TSI-6K1 inverter. When I try to log in with my credentials on the Solar Panels app, it cannot find any devices.
Is it necessary to purchase the Premium Tigo to have access to the API?
[image]

MaikelNL, 2025-01-05:
Same for me. Tigo installation, no new devices found (but them in dutch. Hope someone can help us with that.

patupatu, 2025-01-08:
Hello @MaikelNL,

I activated the free month of EI Premium, and then the Solar Panels app was able to find the device.

The device provides the "instantaneous" production and the daily production.

Too bad that the update of the "instantaneous" production is often very late and some days it can't even get it from the Tigo servers.

The daily production is quite reliable both in terms of consistency and values.

There is no indication of the charge level of the storage, in case I had one, neither in percentage nor in accumulated energy.

I tried to start writing some scripts to interface with the Tigo APIs and get the information I need, but in the meantime my free month expired.

I also contacted Tigo support, who told me that it is possible to connect an energy meter via the rs485 interface via Modbus protocol to the CCA Tigo.

Now I am at a crossroads, whether to invest in contraptions that I have little expertise in to read data directly from the CCA or continue the scripting road to the Tigo API and reluctantly renew the Tigo EI Premium.

I hope my little information has been useful to you.